DanteWinters/lux-power-distribution-card #9: Add indication for grid status

DanteWinters, 2023-04-23:
Add indication for grid status to the card. This can either be that the grid's icon changes on the card, or that a small indicator is added below the grid icon.

guybw, 2023-06-20:
Hey @DanteWinters I'm loving this card FYI. 
I would suggest maybe if you could put a Status at the top left / right just to say "Status: Good" unless it's not? maybe? You can see in my code all the status codes and we should be able to easily work out which is good and which is bad?

DanteWinters, 2023-06-20:
Hi @guybw  Thanks for the feedback!
Using the status codes is a great idea. I want to make release 0.3.0 tonight that has the grid indication in a different way (an icon under the grid image or the grid image can be dimmer). This will work on the grid voltage and the idea is to be usable for people who don't use your lux integration.
I didn't even think of using the status codes but when this release is done (it also has a few other improvements), that's be a high priority for me

DanteWinters, 2023-10-25:
Card has 2 different grid indicator options for showing if the grid is unavailable, and status codes are being displayed. The status code for when grid is unavailable, can be set to show as a warning or as normal behaviour.
The status codes also work with parallel cards.

closed 2023-10-25 by DanteWinters