DanteWinters/lux-power-distribution-card #23: The problem of adding to HA

Xach-s-usami, 2023-06-29:
I followed the instructions, but when adding it, it says in the panel that the card was not found (I attach a screenshot)
Tried through it did not help, the same thing writes that it is not available
[image]
[image]

DanteWinters, 2023-06-30:
I think you manual resource is wrong. It has backslashes in the front and forward-slashes later on in the path.
Remove it and just rely on the HACS path and see if that fixes it.

DanteWinters, 2023-07-05:
@Xach-s-usami  have you been able to resolve the issue?

Xach-s-usami, 2023-07-05:
> @Xach-s-usami  have you been able to resolve the issue?

No, I'm still puzzling over how to solve this case, I deleted everything and did it in stages

Pieterjvr, 2023-07-08:
@Xach-s-usami , It seems like from time to time HACS frontend stop to load. What I have done is to copy the card out and directly under /config/www/ and added it manual.

/local/lux-power-distribution-card.js

[image]

Xach-s-usami, 2023-07-08:
thank you for your help, the problem turned out to be that my sensors are registered with a serial number (but this is not accurate)
I deleted everything, cleared the cache, and then reloaded via Hass
And began to manually write the lines
and I got it all thanks for the help)))
I sit and admire
[image]

closed 2023-07-08 by Xach-s-usami