DanteWinters/lux-power-distribution-card #29: Lux fail status codes

saxon-viking, 2023-07-20:
Hi Dante,

16 = discharging, it isn't a fail

I've deleted that line and I now have normal. I'll update this tomorrow with the full status codes for you.
Cheers

DanteWinters, 2023-07-21:
Hi @saxon-viking ,
The next update will have the status codes be programmed in with a message. Feel free to drop a comment in the discussion  #27 . I'll update the wiki to not use 16 as an example.

closed 2023-07-21 by DanteWinters

saxon-viking, 2023-07-21:
Hi Dante,Github updated. Other LCD codes attached.Hope that helps,CheersHoppySent using the mobile mail appOn 21/07/2023 at 19:43, Dante Winters wrote:
                
            
            
                
                    From: "Dante Winters" ***@***.***>Date: 21 July 2023To: "DanteWinters/lux-power-distribution-card" ***@***.***>Cc: "saxon-viking" ***@***.***>,"Mention" ***@***.***>Subject: Re: [DanteWinters/lux-power-distribution-card] Lux fail status codes (Issue #29)
                
                
Hi @saxon-viking ,
The next update will have the status codes be programmed in with a message. Feel free to drop a comment in the discussion  #27 . I'll update the wiki to not use 16 as an example.