DanteWinters/lux-power-distribution-card #32: Lux card displaying seemingly incorrect inverter/card status - which should be normal?

CaptainMidnight, 2023-08-20:
Just came across this inverter/card status - which isn't actually correct - this is just normal operation for night i.e. no pv input, battery storage powering the house?

[image]

DanteWinters, 2023-08-21:
The statuses are getting reworked. Will be out with the next release.
Currently, you can define which of the status codes that will give it a warning by adding them to the list. I have been told that the example code I gave in the wiki is wrong, but i haven't updated it because of the fact that the feature is getting a rework.

DanteWinters, 2023-08-30:
v1.0.0 is available and this uses the descriptions from the inverter integration. Please be careful when upgrading to the new version, the config changed to allow for parallel inverters and it will break the cards.

CaptainMidnight, 2023-08-31:
Cool beans my friend, I've updated the install on my Has OS box and will be able to see later own when the sun goes down how it handles things. Love the 'energy allocations' section as well whoop whoop!!