DanteWinters/lux-power-distribution-card #35: Get a entity is undefined error

Richspr, 2023-09-08:
Hi I can't seem to get this new code working 
the code for the card is below: I have taken out the dongle number.
I only have one inverter.
In the previous card I didn't use the energy allocations so have not been including that code when setting up the new one.
But still get the error the entity is undefined.
Not an expert on code so any suggestion would be appreciated. 
Thanks Richard

type: custom:lux-power-distribution-card
inverter_count: 1
battery_soc:
  entities:
    - sensor.lux_battery
battery_flow:
  entities:
    - sensor.lux_battery_flow_live
home_consumption:
  entities:
    - sensor.lux_home_consumption_live
grid_flow:
  entities:
    - sensor.lux_grid_flow_live
lux_dongle:
  - BAxxxxxxx 
inverter_alias:
  - Inverter 1
battery_voltage:
  entities:
    - sensor.lux_battery_voltage_live
pv_power:
  entities:
    - sensor.lux_solar_output_live
backup_power:
  entities:
    - sensor.lux_power_to_eps_live
grid_voltage:
  entities:
    - sensor.lux_grid_voltage_live
update_time:
  entities:
    - sensor.lux_data_received_time
  show_last_update: true
grid_indicator:
  hue: true
status_codes:
  entities:
    - sensor.lux_status
  no_grid_is_warning: true
refresh_button: right
energy_allocations:
  entities:
    - sensor.power_plug_1
    - sensor.power_plug_2
    - sensor.power_plug_3
    - sensor.power_plug_4

DanteWinters, 2023-09-09:
Hi @Richspr 
Can you please send a screenshot of the config? I just want make the formatting is correct as these configs are sensitive to indentation.

Unfortunately the new version of the card is sensitive to everything not being perfect, I'm working on a fix to be more description when it encounters an issue

Richspr, 2023-09-09:
Hi DW, thanks for the reply, can I clarify which is the config you want my to send. 
R.

Richspr, 2023-09-09:
HI DW attached should be a screen shot of the config 

[image]

DanteWinters, 2023-09-09:
You said you weren't using the energy allocations, right? Then you need to remove the energy_allocations config and the entities. Remove lines 42 to 47. Let me know if the error still persists.

Richspr, 2023-09-09:
Hi DW, I did that try that but will give it another go, and see if it solves the issue. Thanks R.

DanteWinters, 2023-09-09:
Try removing it an clearing the browser cache? Hit F12 while the browser is open, right-click on the refresh button and it should open a menu where you can select clear cache and hard reload. If you right-click on the refresh button but no options show up, just click on refresh once and try right-clicking again

Richspr, 2023-09-10:
HI DW, thanks for the info, please to report that we are now seeing the card.
Thanks again for your time
Regards Richard

closed 2023-09-10 by DanteWinters