DanteWinters/lux-power-distribution-card #36: Bug in .js file

brickatius, 2023-09-22:
There is a bug in the lux-power-distribution-card.js file which causes this:
Uncaught DOMException: Failed to execute 'define' on 'CustomElementRegistry': this name has already been used with this registry at.......
This is caused by line 433: customElements.define("lux-power-distribution-card", LuxPowerDistributionCard);
As a result, after clearing the browser history the above error is thrown and the card is not rendered properly - just shows the .yaml entry and a message about needing to define the first entity. 
I found that replacing line 433 with the following solves the issue. 
= Object.assign(Object.create(Object.getPrototypeOf(customElements)).define, customElements);
customElements.define = (name, element) => customElements.get(name) || customElements.defineclone(name, element);

DanteWinters, 2023-09-23:
Hi @brickatius. That's a very strange error that you got there, I haven't seen it before. Thanks for finding a solution, I'll test the change and if all is well, I'll add it to the code.
With the error being about the name already being used, I think it might be that there is some remnants of the card somewhere? Can it be that you used the card first time before I implemented with HACS, and then added again with HACS?

Thank you for being diligent in testing with multiple browsers. With Safari, is it native on a Mac or is it Safari on Windows (not sure if it is available for Linux so I'm making an assumption)

brickatius, 2023-09-23:
Hi.I’ve only very recently installed the card. I’m pretty sure I did it through HACS. Safari was on a Mac. I’d did try Firefox on the Mac too, with the same result. The other Firefox was running on a different machine with Ubuntu. (I try to avoid Windows wherever I can.) I’ve tried to replicate the error with the different code on both machines and so far it hasn’t recurred.Regards. Richard. Mob 07973 738655