DanteWinters/lux-power-distribution-card #40: After upgrade to v1.1.1 card broken - "cef.getEntitiesNumState is not a function"

CaptainMidnight, 2023-10-03:
Hi, all I did was update your card and it's broken.... just in case it was my config, I re-used yours from your github page.

Serial blanked to protect the innocent ;o)

```
type: custom:lux-power-distribution-card
inverter_count: 1
parallel:
  average_voltage: true
  parallel_first: true
battery_soc:
  entities:
    - sensor.lux_battery
battery_flow:
  entities:
    - sensor.lux_battery_flow_live
home_consumption:
  entities:
    - sensor.lux_home_consumption_live
grid_flow:
  entities:
    - sensor.lux_grid_flow_live
lux_dongle:
  - BA00000000
inverter_alias:
  - Inverter 1
battery_voltage:
  entities:
    - sensor.lux_battery_voltage_live
pv_power:
  entities:
    - sensor.lux_solar_output_live
backup_power:
  entities:
    - sensor.lux_power_to_eps_live
grid_voltage:
  entities:
    - sensor.lux_grid_voltage_live
update_time:
  entities:
    - sensor.lux_data_received_time
  show_last_update: true
grid_indicator:
  hue: true
status_codes:
  entities:
    - sensor.lux_status
  no_grid_is_warning: true
refresh_button: right
```

DanteWinters, 2023-10-03:
Hi @CaptainMidnight 
Are you installing and upgrading through HACS? If you aren't check that you have all the right files.

Below is something that is added to the readme on my local machine, but not yet committed. Please try this and let me know if the problem persists

### Android

1. Find *Home Assistant* in the list of apps in settings.
2. This step may differ depending on the Android device. Find anything that indicates data used or storage.
3. When there, find the option to clear all the data (cache and storage). Clearning this will log you out of the app and you'll need to log in again.
4. Card should then show up. If it doesn't, please log a bug.

### Web browser

1. With the page open, open the developer console on the browser. Usually it's *F12*.
2. Click on the refresh button.
3. Rick-click on the refresh button to open a menu.
4. Choose the option *Empty Cache and Hard Reload*, or the option closest to this description.

CaptainMidnight, 2023-10-03:
Yes, I used the upgrade via HACS, which broke. Then to be doubly sure, I manually uploaded the latest files directly to the correct www folder and rebooted the HA server.

CaptainMidnight, 2023-10-03:
> Hi @CaptainMidnight Are you installing and upgrading through HACS? If you aren't check that you have all the right files.
>
> Below is something that is added to the readme on my local machine, but not [...]

Yes, that was it, sorry all working now

travismailsa, 2023-10-11:
Just upgraded my HA, Files and card and receiving the error attached even though i removed those plug sensors
[image]

Stewbo1, 2024-08-04:
I also tested with your suggested config data (just changed the inverter ID) and same issue. It's also causing the dashboard to not load first time, so you have to refresh, then the dashboard loads but with either no lux card or the above error. 

Let me know if I can provide anything else. I will move back to the Power Wheel Card for the moment until we've figured this one out so no rush.

Stewbo1, 2024-08-04:
I will try updating Guy's Lux integration now as I haven't in 18 months or so.

Stewbo1, 2024-08-04:
Ok this fixed it on Chrome. I need to work out how to clear the HA cache (log out and back in didn't work) on our iOS devices as well.

Web browser
With the page open, open the developer console on the browser. Usually it's F12.
Click on the refresh button.
Rick-click on the refresh button to open a menu.
Choose the option Empty Cache and Hard Reload, or the option closest to this description.

closed 2024-08-12 by DanteWinters