DanteWinters/lux-power-distribution-card #43: Feature request - display inverter temps

CaptainMidnight, 2023-10-14:
Hi

I was just wondering if it would be possible for you to include displaying the inverter temperatures in the area somewhere next to the inverter?

Ideally:-
sensor.lux_radiator_1_temperature_live
sensor.lux_internal_temperature_live

I may have to look at coding it myself..... just thought I'd ask before going down a potential rabbit hole...

DanteWinters, 2023-10-14:
Hi. We can look at it, my reservation is simply the amount of stuff on the card. My original goal was to make a card that resembles the layout on the LuxPower app, and add a few features. There are a lot of stats that would be nice, but can make the card busy.
I'm happy to look at it, but there are other features that I want to implement instead. If you are willing to take a look and implement it bulk of the code, that would be a great help.

CaptainMidnight, 2023-10-14:
Yeah no problems, I'll have a look at the code and take it from there.

DanteWinters, 2024-08-02:
With the release of v1.5.0, the option is there to show the individual PV array values. The change made me think that I can eventually also add the individual voltages. Since there is more info that might eventually be added, I want to keep this one open as I might add it at some point

DanteWinters, 2025-02-10:
Hi @CaptainMidnight 
Added the option for temps with the last update. The card editor isn't working, but you can just add them to the config:
```yaml
temp:
  entities:
    - sensor.lux_internal_temperature_live
```