DanteWinters/lux-power-distribution-card #57: undefined is not an object (evaluating this._config.generator_power.is_used)

Stewbo1, 2024-08-03:
Hi there,
Great card thank you, i use it with Guy’s LuxPower inverter integration.
I ran a few updates today, core HA and in HACS and since then I have been getting this error.
I cannot find the variable in the card’s settings so I am a bit lost. Thanks

undefined is not an object (evaluating this._config.generator_power.is_used)

DanteWinters, 2024-08-03:
Hi. After updates when I change the config, the card always acts up a little. Please check issue #40 and see if that fixes it, otherwise I'll see about the fix

Stewbo1, 2024-08-04:
Thank you. I followed those steps to clear cache (well on IoS so logged out/back in) and I still have it. Either the error or a partial status image which doesn’t show the output power to home/grid. I’ll move over now to #40 and provide a screenshot when I get a mo.

DanteWinters, 2024-08-04:
please check on another device or website if you still have the issue. If the issue is still visible on web, then it's a config or card issue. If not, then it's a cache issue

Stewbo1, 2024-08-04:
Yes on Chrome on my PC I get "Cannot read properties of undefined (reading 'is_used')

Stewbo1, 2024-08-04:
Fixed by doing the forced cache clearance on all devices one by one.

closed 2024-08-04 by Stewbo1