DanteWinters/lux-power-distribution-card #62: AC coupling icon request

Rumast22, 2024-08-18:
First off, great work!

Would it be possible to add an AC coupling/generator icon for - sensor.lux_generator_power_live?  I have a microinverter feeding into my aux input that I like to see, which I can in the Luxpower/EG4 app.

I'm checking using the allocated power to do that but it feeds into home rather than the inverter.

DanteWinters, 2024-08-21:
Hi @Rumast22 
I added the image that you see on the website for generator to the card on the last update. When I updated the logic behind the card, I made it so that the generator icon will show when there is a positive value on sensor.lux_generator_power_live, but no flow of grid.
I don't fully think this is the right logic. I saw that the integration makes no distinction between Grid A (Primary) and Grid B (Generator). So If you are using Grid B, Grid A still gives value. Which I think is a mistake. I haven't spent time on the integration with the generator values (I don't have one but a friend does so I saw some of the values) so I decided to wait a little before updating the card. I'll see if I have time before the end of the month to try something there, otherwise I'll update the cards logic so that you can use it.

seanocali, 2025-02-16:
At my home some of my panels are AC coupled and some are PV input strings. It would be great if we could just add generator input power as an extra solar source and it would combine the wattage of the two.

DanteWinters, 2025-02-17:
Hi @seanocali 
What inverter do you have and what integration are you working with?
The problem with the generator input is balancing it with grid. If it's a case that either the inverter uses grid or generator, then it's easier because in code it's a simple thing of either or, but when it can pull from grid and the generator, then unfortunately it is beyond the scope of this card. 
I like the solution you presented, I think it will be a good option that shows power in without affecting the card too much. I'll see what options I can come up with for generator inputs to get a better layout of the info.