DanteWinters/lux-power-distribution-card #63: arrows from grid wrong  direction

prspyder, 2024-10-23:
Hi so I dont know if by default is like this but if I'm exporting power to the grid arrows should be pointed at the other direction and it seems mine is working backwards.  

[image]
[image]

danielraffel, 2024-11-15:
I haven't experienced this since I am not exporting power yet (tho it does occasionally seem to push tiny amounts and the arrows appear to be correct.) For the sake of others who might stumble on this issue can you share the helper and anything else you had to do? I presume the configuration.yaml change might be something like:

```
template:
  - sensor:
      - name: "Inverted Grid Power"
        state: >
          {{ states('sensor.grid_power') | float * -1 }}
        unit_of_measurement: "W"
        unique_id: "inverted_grid_power"
        icon: "mdi:power-plug-off"
```

DanteWinters, 2025-02-10:
Hi @prspyder 
Based on the name of the entity you have (Deye/SunSync/Sol-Ark Grid power), I'm going to assume you aren't using this card with a LuxPower inverter? The arrows are fine for the LuxPower integration. I think the integrations just have it differently. Nevertheless, I added an optional config you can add to your card to reverse the grid flow card.
```yaml
grid_flow:
  entities:
    - sensor.your_entity
  reverse: true
```
This should fix it without needing to use a template. Implementing it on my system shows the desired result

prspyder, 2025-02-12:
well just wanted to let you know that I eliminated the helper changed the sensor to the correct one that should work and its broken now so I gave up on using this card for me. However I haven't updated my dads one and that one keeps working without issue  is there a way you could make the card appear in custom cards?

DanteWinters, 2025-02-16:
Latest update should now show up on the list of custom cards. The editor also works on all parts, but does give errors when you first create the card (because of missing entities). Feel free to test again. If you are having issues after updating, please refer to #61 and see if that fixes the issue.

closed 2025-10-21 by DanteWinters