DanteWinters/lux-power-distribution-card #64: Smart Load Support Inquiry

danielraffel, 2024-11-15:
I’m about to connect my EV charger to the generator port and enable smart load. Do you happen to know whether the card supports such a setup and how this might impact visual reporting? Are there any specific tests or feedback that would be helpful once it’s connected? Fantastic work—your card is greatly appreciated! I wrote about how I'm using it [here](https://danielraffel.me/2024/10/28/how-i-built-a-real-time-solar-dashboard-for-my-eg4-18kpv-inverter-using-home-assistant/).

closed 2025-02-12 by danielraffel

DanteWinters, 2025-02-12:
Hi @danielraffel 
I'm glad you were able to close the ticket. I know you made it last year, but the generator just wasn't a priority for me to implement. I believe the implementation I have is simply to check if generator power is more than 0 but grid is zero, then it shows generator.
The integration this was built for, doesn't really do generator input, and I've noticed people with inverters other than Luxpower use the card. My plan is to have a setting for the generator: `match`, `replace` and `override` are the current ones.
 - `match` means if the generator input is equal to the grid flow, I show generator. This setting is because on the last tests we had with the luxpower, it showed that when a generator is on, grid1 and 2 have the same value
 - `replace` is what one would expect. If there is generator but no grid, use grid.
 - `override` is just something that I'm adding as a backup for some unforeseen option: if generator power is more than 0, use it.

I hope the card works well for you and if you want to improve on the codebase you are welcome to submit a PR.

danielraffel, 2025-02-12:
Thanks Dante! I learned quite a bit about smart load (on my specific 18kpv inverter) and discovered it's buggy and not so smart so I did not end up using it. From what I learned it appears that it might have been a software release that was enabled after the hardware shipped without deep consideration for some of the hardware requirements to properly support it. 

I regularly look at this card and am really appreciative of the time you put in to build, share and support this. I've debated making a very basic iOS app for myself using live activities so that I can see near realtime generation "at a glance" as a widget on my phone / watch but haven't gotten around to it and not sure if I will!