DanteWinters/lux-power-distribution-card #71: Popup Charts No Longer Accessible in v1.6.0+

danielraffel, 2025-02-13:
In version 1.5.1 and earlier, when viewing the power flow card, hovering over and tapping icons like the battery would display graphs.

[image]

[image]

However, in version 1.6.0 and later, tapping on these icons no longer brings up the graphs, and the links to the popup charts seem to not be working for me on web, desktop or native app. 

Is anyone else experiencing this issue? Did something change? Any chance of a fix?

gecomaia, 2025-02-14:
Same behavior on my side, exactly as you’ve described!

DanteWinters, 2025-02-15:
I found this after updating. In truth, I was hoping it wasn't a useful enough feature that people would notice, or complain. I am pleased to be wrong. Investigating, fix will hopefully be out this weekend

danielraffel, 2025-02-15:
Thank you @DanteWinters 🙇

DanteWinters, 2025-02-15:
Issue found and resolved. Going to fix issue in discussion #65 before making release. Same ETA for update.

DanteWinters, 2025-02-16:
Updated. Please close when you have tested it.

danielraffel, 2025-02-16:
Works great here thanks!!

closed 2025-02-16 by danielraffel

gecomaia, 2025-02-16:
Thanks, it's working perfectly again!