DanteWinters/lux-power-distribution-card #73: Issue with version 1.6 onwards

Drewskieza, 2025-02-17:
If I upgrade to any 1.6 version or above I get thi error but if I roll back to 1.5.1 it seems to work fine.

[image]

Am I missing something wiht the new update?

DanteWinters, 2025-02-17:
Please provide your config, excluding sensitive info like your dongle serial number.
Are you upgrading to 1.6.2, and have you tried the steps outline in issue #61 ?

Lepais01, 2025-02-17:
I have the same issue showed by Drewskieza with 1.6.2. In addition under Register hereafter the error:

"Uncaught error from Chrome 133.0.0.0 on Windows 10 TypeError: hf.generateTemp is not a function LuxPowerDistributionCard.createCard (/hacsfiles/lux-power-distribution-card/lux-power-distribution-card.js:57:52) set hass (/hacsfiles/lux-power-distribution-card/lux-power-distribution-card.js:16:12)"

Drewskieza, 2025-02-17:
Thanks, told you I missed something. I was sure I had followed the steps but obviously not.

DanteWinters, 2025-02-17:
Hi @Lepais01 can you please share your config and have you followed the steps in #61 ? 

Hi @Drewskieza I'm not sure with you last message, did the issue get resolved when you cleared the cache?

Drewskieza, 2025-02-17:
> Hi [@Drewskieza](https://github.com/Drewskieza) I'm not sure with you last message, did the issue get resolved when you cleared the cache?

Yes, issue is resolved and all working as expected.

Lepais01, 2025-02-17:
@DanteWinters  solved appyling step in #61 , Thanks!

closed 2025-02-17 by Drewskieza