DanteWinters/lux-power-distribution-card #76: Card no longer working - permissions issue?

JaseCat, 2025-03-15:
Before you log a bug, please read through (IN FULL, AND FOLLOW THE INSTRUCTIONS) of the following bug: #61 

If you have done that and the problem still persists, please fill in the following report:

# Bug report
 - I have looked at issue #61 and followed the steps: Yes
 - The last version that worked fine for me was: 1.6.2
 - I'm trying to upgrade to version: n/a
 - I have tested it on the following devices/browsers: Apple iPhone/ Android tablet/ Apple tablet/ Edge browser/ Safari browser/

## Config:

Please copy your config below while removing all sensitive information

type: custom:lux-power-distribution-card
inverter_count: 1
parallel:
  average_voltage: true
  parallel_first: true
battery_soc:
  entities:
    - sensor.lux_battery
battery_flow:
  entities:
    - sensor.lux_battery_flow_live
home_consumption:
  entities:
    - sensor.lux_new_home_consumption_live
grid_flow:
  entities:
    - sensor.lux_grid_flow_live
lux_dongle:
  - ********
inverter_alias:
  - Inverter 1
battery_voltage:
  entities:
    - sensor.lux_battery_voltage_live
pv_power:
  entities:
    - sensor.lux_solar_output_live
backup_power:
  entities:
    - sensor.lux_power_to_eps_live
grid_voltage:
  entities:
    - sensor.lux_grid_voltage_live
update_time:
  entities:
    - sensor.lux_data_received_time
  show_last_update: true
grid_indicator:
  hue: true
status_codes:
  entities:
    - sensor.lux_status
  no_grid_is_warning: true
refresh_button: right

###  Card error

The following is the error the card gives on Home Assistant:

[image]

### Browser console

When on a browser, the following error is given in the console log:

[image]

### Additional info

(Any other useful info?)

greytuk, 2025-03-15:
The file lux-power-distribution-card.js is making a call to https://unpkg.com/lit-element@2.0.1/lit-element.js?module

unpkg.com has been down in multiple regions for over 12 hours so far....

jdvdm, 2025-03-15:
Also not working on my side

DanteWinters, 2025-03-15:
I see the issue. Unfortunately, I'm not a JavaScript developer and have no idea how to fix this. Based on @greytuk 's comment, this issue will hopefully resolve itself soon.
One of my friends is a JavaScript developer, so I'll ask him to take a look and see if he can find a solution that also makes that this won't happen again. He won't be able to look at it until after the weekend

chautruongthinh, 2025-03-15:
> I see the issue. Unfortunately, I'm not a JavaScript developer and have no idea how to fix this. Based on [@greytuk](https://github.com/greytuk) 's comment, this issue will hopefully resolve itself [...]

We can use local file to fix it. You can see here: https://github.com/AmoebeLabs/flex-horseshoe-card/issues/88#issuecomment-2145334978

greytuk, 2025-03-15:
My temporary fix is 

- Delete .gz files at config\www\community\lux-power-distribution-card
- Edit lux-power-distribution-card-editor.js ands replace https://unpkg.com/lit-element@2.0.1/lit-element.js?module with https://esm.run/lit-element@2.0.1/lit-element.js?module
- Go to Hacs and update information for the Lux Power Card
- Go to Settings\Dashboard\ hit the 3 dots and select resources
- edit  luxpower card and incremenent the HacsID. for example /hacsfiles/lux-power-distribution-card/lux-power-distribution-card.js?hacstag=631525923171

Test from a Windows web browser. Press Shift F5 to refresh the cache. You may need to clear the cache as well

From a HA IOS app go to settings\companion app\debugging and click Reset Frontend cache

Some of the above steps are overkill and probably not needed, but this is the sequence that finally worked for me

[image]

DanteWinters, 2025-03-15:
Update to V1.6.3 to fix. Thanks @greytuk 
I'll get a better solution for this

mariusl, 2025-03-15:
I see HACS has not been updated with 1.6.3 yet. How long does it take usually?

danielraffel, 2025-03-15:
@DanteWinters thank you for being such a wonderful steward and maintainer. We owe you. 🙇

DanteWinters, 2025-07-10:
@QuantumRipple 
Working on a TypeScript rebuild of the card. Will have a single exported file that has all the required resources, like most other dashboard cards have. ETA: Unknown